FemmeVerbeek/Arduino_LSM9DS1 #7: readRawAccel - Doesnt provide raw data

rishikanths, 2020-06-18:
Hi,

When I try to access the raw data using the following command,
  IMU.readRawAccel(gx,gy,gz); 

and calibrated data using 
    IMU.readAccel(x,y,z);
However, both the data values are the same?

FemmeVerbeek, 2020-06-18:
Hi rishikanths,
Did you calibrate the Accelerometer? 
If not, this is precisely as expected. The problem is that no two LSM9DS1 chips are the same. In order to get accurate output you must find the calibration values for your instance of the chip. 
There is a link to a Youtube video in the readme that show what to do for a calibration.
https://youtu.be/BLvYFXoP33o

rishikanths, 2020-06-20:
I am trying to use the DIY calibration, but facing issues calibrating gyro. Due to memory issues, I had to break the DIY sketch into three parts - aceel, gyro and magno. I dont think that will be an issue, but correct me if I am wrong. 

During the gyro callibration- 

1. I press "A" to provide a angle. However, the program doesnt wait for my input. I have to press"a" and quickly give my angle ("90"). If I take a few seconds, the program just executes without taking any values. The issue might be with this line of code ` while (!Serial.available());` 

2. Even If I am successful in providing angle, the function `calibrateGyroslope` doesnt execute as expected. I always get 
> Measured direction change X 0.000000°	Y 0.000000°	 Z 0.000000°
> Std.dev.      NAN	     NAN	nan
> Not a valid measurement!

Irrespective of how I turn the sensor. Again I think its this ` while (!Serial.available());`  which is causing the issues. I doesnt enter the while loop.

Can you shed some light on this. I am using LSM9DS1 on UNO.

Thanks

rishikanths, 2020-06-20:
Thanks for the help. I was able to use DIY calibration sketch. However, I had to break it into three due to space constraints. I wrote a small sketch to compare outputs using your library and standard "Adafruit_LSM9DS1.h". The following is the output (first three values are using `IMU.readRawAccel(gx,gy,gz);` and other three are `lsm.getEvent(&a, &m, &g, &temp);` where `Adafruit_LSM9DS1 lsm = Adafruit_LSM9DS1();` 

```
-0.0483, -0.0165, 0.9963;-0.47,-0.16,9.77
-0.0479, -0.0146, 0.9988;-0.47,-0.14,9.79
-0.0471, -0.0148, 0.9996;-0.46,-0.15,9.80
-0.0496, -0.0118, 0.9978;-0.49,-0.12,9.78
-0.0455, -0.0156, 0.9896;-0.45,-0.15,9.70
-0.0488, -0.0135, 0.9967;-0.48,-0.13,9.77
....
```

As shown, the values using your library (first three) differ by a factor of 10 when compared to standard library. 

Am I missing something in the sketch?

FemmeVerbeek, 2020-06-23:
The reason is that Adafruit gives the acceleration in m/s2 and readRawAccel in g  (gravity).
1g = 9.81 m/s2
If you want to see the same output you must start your program with 
IMU.accelUnit=METERPERS2;
and use IMU.readAccel instead of IMU.readRawAccel.  The latter is only a recent addition for the purpose of writing your own calibration procedure.

FemmeVerbeek, 2020-06-23:
I saw some comments in the mail about the program not waiting for keyboard input, but I don't see it here. I guess you solved it?

By the way I like the fact that you are using a UNO and a separate LSM9DS1.
I'm curious if some of the defects in the chip also exist on your configuration.
Would you be willing to run the RegisterTest sketch and post the output here?.

rishikanths, 2020-06-24:
Testing all library chip settings

FunetODR sample rate result
 Hz 
 Hz 
Testing all library chip settings

Full Scale settings should hardly inluence measured values
setAccelFS(0 g Accel uncalibrated g g g 
setAccelFS(1 g Accel uncalibrated g g g 
setAccelFS(2 g Accel uncalibrated g g g 
setAccelFS(3 g Accel uncalibrated g g g 
setAccelFS parameter out of range 

 deg/s Gyro raw dps dps dps 
Gyro calibrated dps dps dps 
 deg/s Gyro raw dps dps dps 
Gyro calibrated dps dps dps 
 deg/s Gyro raw dps dps dps 
Gyro calibrated dps dps dps 
 deg/s Gyro raw dps dps dps 
Gyro calibrated dps dps dps 
setGyroFS full scale parameter out of range 

 µT Magnet µT µT µT 
 µT Magnet µT µT µT 
 µT Magnet µT µT µT 
 µT Magnet µT µT µT 
setMagnetFS full scale parameter out of range 

Test Operational Mode 0=off, 1 accelerometer only, 2= Accel + Gyro 

Default operational mode 2
setGyroODR(0) Operational mode 1
setAccelODR(0) Operational mode 0
setGyroODR(3) Operational mode 2

 setGyroODR result (shared ODR, i>0 overrules the Accel setting)
Settting 0 Oper.mode 1 ODR Gyro 0.000Hz Accel 116.279Hz
Settting 1 Oper.mode 2 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 2 Oper.mode 2 ODR Gyro 57.248Hz Accel 57.248Hz
Settting 3 Oper.mode 2 ODR Gyro 113.250Hz Accel 113.250Hz
Settting 4 Oper.mode 2 ODR Gyro 228.854Hz Accel 228.854Hz
Settting 5 Oper.mode 2 ODR Gyro 453.571Hz Accel 453.571Hz
Settting 6 Oper.mode 2 ODR Gyro 500.751Hz Accel 500.751Hz
setGyroODR parameter out of range 7

 setAccelODR result (shared ODR, i=0 switches off Gyro and Accel)
Settting 0 Oper.mode 0 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 1 Oper.mode 2 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 2 Oper.mode 2 ODR Gyro 57.274Hz Accel 57.274Hz
Settting 3 Oper.mode 2 ODR Gyro 113.379Hz Accel 113.379Hz
Settting 4 Oper.mode 2 ODR Gyro 228.938Hz Accel 228.938Hz
Settting 5 Oper.mode 2 ODR Gyro 462.030Hz Accel 462.030Hz
Settting 6 Oper.mode 2 ODR Gyro 490.458Hz Accel 490.458Hz
setAccelODR parameter out of range 7

 Switch off Gyro (Accelerometer only mode)
Settting 0 Oper.mode 0 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 1 Oper.mode 1 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 2 Oper.mode 1 ODR Gyro 0.000Hz Accel 49.525Hz
Settting 3 Oper.mode 1 ODR Gyro 0.000Hz Accel 116.198Hz
Settting 4 Oper.mode 1 ODR Gyro 0.000Hz Accel 233.863Hz
Settting 5 Oper.mode 1 ODR Gyro 0.000Hz Accel 462.030Hz
Settting 6 Oper.mode 1 ODR Gyro 0.000Hz Accel 501.086Hz
setAccelODR parameter out of range 7

 setMagnetODR sample rate result
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
 Hz 
setMagnetODR  parameter out of range 16

FemmeVerbeek, 2020-06-24:
Thx for your trouble and no this is definitely not what I expected.
The ODR data is interesting. Setting 6 should have resulted in 952 Hz, but it did not on both our systems. So it is truly an anomaly in the documentation. 
It looks like the UNO does not handle the String object well, and I used that a lot in the programs.
This is also why changing a value in the calibration program did not work. 

Can I ask for your help again?

Attached the revised test program which I think will run. 

I did not have time yet to split the calibration program in three.

[LSM9DS1_RegisterTest.zip](https://github.com/FemmeVerbeek/Arduino_LSM9DS1/files/4828250/LSM9DS1_RegisterTest.zip)

rishikanths, 2020-07-05:
Sorry for the delayed response

Testing all library chip settings

Full Scale settings should hardly inluence measured values

setAccelFS(0) Setting 2.00 g Accel uncalibrated -0.04g 0.01g 1.01g 
setAccelFS(1) Setting 24.00 g Accel uncalibrated -0.03g 0.00g 1.01g 
setAccelFS(2) Setting 4.00 g Accel uncalibrated -0.04g 0.01g 1.00g 
setAccelFS(3) Setting 8.00 g Accel uncalibrated -0.04g 0.00g 1.00g 
setAccelFS parameter out of range 

Gyro Calibration @FS 500 dps
setGyroFS(0) Setting 245.00 deg/s Gyro raw -0.50dps 0.15dps -0.69dps Gyro calibrated -0.36dps -0.12dps -0.21dps 
setGyroFS(1) Setting 500.00 deg/s Gyro raw -0.17dps 0.29dps -0.48dps Gyro calibrated -0.04dps -0.02dps -0.03dps 
setGyroFS(2) Setting 1000.00 deg/s Gyro raw 0.52dps 0.80dps -0.25dps Gyro calibrated 0.67dps 0.49dps 0.25dps 
setGyroFS(3) Setting 2000.00 deg/s Gyro raw 0.38dps 1.06dps -0.42dps Gyro calibrated 0.49dps 0.74dps 0.03dps 
setGyroFS parameter out of range 

setMagnetFS(0) Setting 400.00 µT Magnet 9.25µT 23.77µT -37.68µT 
setMagnetFS(1) Setting 800.00 µT Magnet 18.24µT 47.66µT -74.66µT 
setMagnetFS(2) Setting 1200.00 µT Magnet 28.10µT 72.07µT -112.29µT 
setMagnetFS(3) Setting 1600.00 µT Magnet 36.92µT 96.44µT -149.59µT 
setMagnetFS parameter out of range

 setGyroODR result (shared ODR, i>0 overrules the Accel setting)
Settting 0 Oper.mode 1 ODR Gyro 0.000Hz Accel 116.387Hz
Settting 1 Oper.mode 2 ODR Gyro 62.158Hz Accel 62.158Hz
Settting 2 Oper.mode 2 ODR Gyro 55.729Hz Accel 55.729Hz
Settting 3 Oper.mode 2 ODR Gyro 113.404Hz Accel 113.404Hz
Settting 4 Oper.mode 2 ODR Gyro 229.274Hz Accel 229.274Hz
Settting 5 Oper.mode 2 ODR Gyro 454.621Hz Accel 454.621Hz
Settting 6 Oper.mode 2 ODR Gyro 501.253Hz Accel 501.253Hz
setGyroODR parameter out of range 7

 setAccelODR result (shared ODR, i=0 switches off Gyro and Accel)
Settting 0 Oper.mode 0 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 1 Oper.mode 2 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 2 Oper.mode 2 ODR Gyro 57.300Hz Accel 57.300Hz
Settting 3 Oper.mode 2 ODR Gyro 113.353Hz Accel 113.353Hz
Settting 4 Oper.mode 2 ODR Gyro 229.106Hz Accel 229.106Hz
Settting 5 Oper.mode 2 ODR Gyro 454.170Hz Accel 454.170Hz
Settting 6 Oper.mode 2 ODR Gyro 501.672Hz Accel 501.672Hz
setAccelODR parameter out of range 7

 Switch off Gyro (Accelerometer only mode)
Settting 0 Oper.mode 0 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 1 Oper.mode 1 ODR Gyro 0.000Hz Accel 0.000Hz
Settting 2 Oper.mode 1 ODR Gyro 0.000Hz Accel 49.446Hz
Settting 3 Oper.mode 1 ODR Gyro 0.000Hz Accel 116.387Hz
Settting 4 Oper.mode 1 ODR Gyro 0.000Hz Accel 234.126Hz
Settting 5 Oper.mode 1 ODR Gyro 0.000Hz Accel 463.119Hz
Settting 6 Oper.mode 1 ODR Gyro 0.000Hz Accel 501.505Hz
setAccelODR parameter out of range 7

 setMagnetODR sample rate result
setMagnetODR(0) Setting 0.00 Hz 
setMagnetODR(1) Setting 0.00 Hz 
setMagnetODR(2) Setting 0.00 Hz 
setMagnetODR(3) Setting 0.00 Hz 
setMagnetODR(4) Setting 0.00 Hz 
setMagnetODR(5) Setting 0.00 Hz 
setMagnetODR(6) Setting 38.98 Hz 
setMagnetODR(7) Setting 77.45 Hz 
setMagnetODR(8) Setting 472.56 Hz 
setMagnetODR(9) Setting 481.61 Hz 
setMagnetODR(10) Setting 485.50 Hz 
setMagnetODR(11) Setting 470.43 Hz 
setMagnetODR(12) Setting 501.50 Hz 
setMagnetODR(13) Setting 486.09 Hz 
setMagnetODR(14) Setting 474.35 Hz 
setMagnetODR(15) Setting 485.48 Hz 
setMagnetODR  parameter out of range 16

 Changing output units 

readAccel -0.36m/s2 0.05m/s2 9.74m/s2 
readAccel -0.04g 0.01g 0.99g 
  Simulate gyro measurement by offsetting 100 dps
readGyro -99.90 deg/s -99.81 deg/s -99.98 deg/s 
readGyro -1.74 rad/s -1.74 rad/s -1.75 rad/s 
readGyro -16.65 RPM -16.65 RPM -16.68 RPM 
readGyro -0.28 RPS -0.28 RPS -0.28 RPS 
readMagnet  37.37 µT 95.06 µT -150.56 µT 
readMagnet  36306.15 nT 94228.52 nT -150917.96 nT 
readMagnet  0.37 G 0.94 G -1.51 G

FemmeVerbeek, 2020-07-07:
Thx for your trouble. 
The AccelFS behavior looks good. According to the datasheet Setting 1 should be 16g but turns out to be 24g.
Themax gyroscope error is of the order 0.7 deg/s. I found a way to improve significantly on this. 
The setMagnetFS does not work, on neither of our set-ups.  I don't think this is due to the library function. It copies the correct value to the indicated register, but the chip does not do anything with it.
New to me is that on your setup Setting 1 of the Accel and Gyro and settings 0 to 5 of setMagnetODR don't work. Probably this is not bad as they are the slowest settings.

I found a way to reduce the memory usage by a very large factor. It turns out that the UNO compiler copies all the Serial.prints to RAM. By means of encapsulating the text in the F(     ) macro it prints directly from flash.   
I'll upload a new version within a few days. The DIY calibration is split into three separate programs to reduce the footprint.

FemmeVerbeek, 2020-07-10:
Hi Rishikanths
The DIY calibration programs are now split into three separate sketches. It should be no problem to run any of them on an Arduino Uno. Compiled on a UNO I have here, the memory usage was close to 50%, so there should be no issues. I do not have a separate LSM9DS1 board, so I can't test that part. 
The FS and ODR settings can be changed in the DIY calibration programs, and they are included in the copy/paste-able code shown onscreen. It turned out that this resulted in a more accurate calibration.
I solved the problem with the setMagnetFS. 

Let me know how it runs on your system now.

Do you have some information for me what you used for testing. What did you use for the LSM9DS1 chip, and how did you connect it to the Uno board?

Regards Femme