FemmeVerbeek/Arduino_LSM9DS1 #10: Accelerometer full scale 16G

MigSanc, 2020-11-12:
In the library you have:

`float LSM9DS1Class::getAccelFS() // Full scale dimensionless, but its value corresponds to g
{   float ranges[] ={2.0, 24.0, 4.0, 8.0}; //g
    uint8_t setting = (readRegister(LSM9DS1_ADDRESS, LSM9DS1_CTRL_REG6_XL) & 0x18) >> 3;
    return ranges[setting] ;
}`

However the range should not be 24g, but rather 16g.
So if one modifies the range to `float ranges[] ={2.0, 16.0, 4.0, 8.0}; //g` and do IMU.setAccelFS(1) the readings become wrong

MigSanc, 2020-11-12:
There is an error.

When calculating the scale for the sensitive one cannot use SCALE/32768. This formula works others for all the others sensitivities on the accelerometer but fails for the 16G.

According to the datasheet:
image

but 16/32768 does not correspond to the sensitity value. However 24g will give that value. I have contacted ST to advise.

FemmeVerbeek, 2020-11-12:
Correct. I noticed that putting 16 as multiplication factor results in a measured gravity value of 0.66..... so 24 is the correct multiplication value. Further with this correction in place the accelerometer maxed out at 20 g, so at this setting it is not using the full 16 bits of to communicate. This is contrary to the other settings, they maxed out excactly at their corresponding FS values.

closed 2021-02-01 by FemmeVerbeek