FemmeVerbeek/Arduino_LSM9DS1 #11: Problem running calibration

cdeerinck, 2020-11-15:
I can get the three "Simple" examples to run fine, but when I try to run any of the three calibration examples, they print the first line to the Serial Monitor, and then they freeze.  If I try to rerun the program a second time, the entire Arduino IDE freezes up.  

When they freeze, I get the 3X long, 3X short blink pattern indicating that the Mbed OS has crashed. If I power the Arduino down, and replug it in, it does the same thing.

when I run  the Register Test, I get the following:

```
Testing all library chip settings

Full Scale settings should hardly inluence measured values

setAccelFS(0) Setting 2.00 g Accel uncalibrated -0.00g -0.21g 0.95g 
setAccelFS(1) Setting 24.00 g Accel uncalibrated -0.00g -0.22g 0.95g 
setAccelFS(2) Setting 4.00 g Accel uncalibrated -0.00g -0.21g 0.96g 
setAccelFS(3) Setting 8.00 g Accel uncalibrated 0.00g -0.21g 0.96g 
setAccelFS parameter out of range 

setGyroODR (4) = 235.39Hz 50 samples
setGyroFS(0) Setting 245.00°/s Gyro raw -0.39°/s -0.02°/s 0.23°/s Gyro calibrated 0.13°/s 0.01°/s 0.02°/s 
setGyroFS(1) Setting 500.00°/s Gyro raw 0.29°/s -0.05°/s -0.47°/s Gyro calibrated -0.07°/s 0.03°/s 0.03°/s 
setGyroFS(2) Setting 1000.00°/s Gyro raw 0.29°/s 0.33°/s -0.06°/s Gyro calibrated -0.07°/s -0.03°/s 0.01°/s 
setGyroFS(3) Setting 2000.00°/s Gyro raw -0.12°/s -0.31°/s -0.11°/s Gyro calibrated -0.10°/s -0.00°/s 0.01°/s 
setGyroFS parameter out of range 

setMagnetFS(0) Setting 400.00 µT Magnet 15.64µT 56.88µT -58.20µT 
setMagnetFS(1) Setting 800.00 µT Magnet 15.87µT 57.52µT -59.13µT 
setMagnetFS(2) Setting 1200.00 µT Magnet 16.16µT 57.69µT -59.56µT 
setMagnetFS(3) Setting 1600.00 µT Magnet 16.69µT 57.92µT -59.69µT 
setMagnetFS parameter out of range

```

I'm pretty sure it is something wrong with my setup, and not your code, but I can't seem to figure out what.  Any ideas?

Also, when compiling, I get the following warnings:

```
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino: In function 'void testAccelFS()':
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:64:65: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       {  printResult ("\nsetAccelFS(", i ,IMU.getAccelFS()," g ");
                                                                 ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:64:65: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:67:55: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
          printXYZData("Accel uncalibrated ",x,y,z,"g ");
                                                       ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:67:55: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino: In function 'void testGyroFS()':
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:78:65: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
          { printResult ("\nsetGyroFS(",i,IMU.getGyroFS(),"°/s ");
                                                                 ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:78:65: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:82:50: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
            printXYZData("Gyro raw ",x,y,z,"°/s ");
                                                  ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:82:50: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:84:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
            printXYZData("Gyro calibrated ",x,y,z,"°/s ");
                                                         ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:84:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino: In function 'void testMagnetFS()':
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:115:69: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       {  printResult ("\nsetMagnetFS(", i ,IMU.getMagnetFS()," µT ");
                                                                     ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:115:69: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:118:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
          printXYZData("Magnet ",x,y,z,"µT ");
                                             ^
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:118:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino: In function 'void testMagnetODR()':
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/examples/RegisterTest/RegisterTest.ino:163:71: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
[... 69 lines cut ...]
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp:491:61: warning: 'startM' may be used uninitialized in this function [-Wmaybe-uninitialized]
     magnetODR= (1000000.0*float(countM)/float(lastEventTimeM-startM) );
                                               ~~~~~~~~~~~~~~^~~~~~~
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp:489:60: warning: 'startA' may be used uninitialized in this function [-Wmaybe-uninitialized]
     accelODR= (1000000.0*float(countA)/float(lastEventTimeA-startA) );
                                              ~~~~~~~~~~~~~~^~~~~~~
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp:491:61: warning: 'lastEventTimeM' may be used uninitialized in this function [-Wmaybe-uninitialized]
     magnetODR= (1000000.0*float(countM)/float(lastEventTimeM-startM) );
                                               ~~~~~~~~~~~~~~^~~~~~~
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp:489:60: warning: 'lastEventTimeA' may be used uninitialized in this function [-Wmaybe-uninitialized]
     accelODR= (1000000.0*float(countA)/float(lastEventTimeA-startA) );
                                              ~~~~~~~~~~~~~~^~~~~~~
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp: In member function 'float LSM9DS1Class::measureMagnetODR(long unsigned int)':
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp:534:54: warning: 'lastEventTime' may be used uninitialized in this function [-Wmaybe-uninitialized]
    return (1000000.0*float(count)/float(lastEventTime-start) );
                                         ~~~~~~~~~~~~~^~~~~~
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp: In member function 'float LSM9DS1Class::measureAccelGyroODR()':
/Users/Chuck/Documents/Arduino/libraries/Arduino_LSM9DS1-master/src/LSM9DS1.cpp:514:52: warning: 'lastEventTime' may be used uninitialized in this function [-Wmaybe-uninitialized]
  return (1000000.0*float(count)/float(lastEventTime-start) );
                                       ~~~~~~~~~~~~~^~~~~~
/Users/Chuck/Library/Arduino15/packages/arduino/hardware/mbed/1.3.0/libraries/Wire/Wire.cpp: In member function 'void arduino::MbedI2C::receiveThd()':
/Users/Chuck/Library/Arduino15/packages/arduino/hardware/mbed/1.3.0/libraries/Wire/Wire.cpp:142:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      if (c <= sizeof(buf)) {
          ~~^~~~~~~~~~
```

cdeerinck, 2020-11-15:
I was playing around with trying to isolate the problem.  It appears as though the problem is when anything tries to print more than 64 bytes in a single print statement.

```
void setup() {
  pinMode(LEDR, OUTPUT);
  pinMode(LEDG, OUTPUT);
  pinMode(LEDB, OUTPUT);
  digitalWrite(LEDR, HIGH);  digitalWrite(LEDG, HIGH);  digitalWrite(LEDB, HIGH);  // turn the LED black 
  Serial.begin(115200); 
    digitalWrite(LEDR, LOW);  digitalWrite(LEDG, HIGH);  digitalWrite(LEDB, HIGH);  // turn the LED red 
  delay(1000);
  while (!Serial);
  Serial.println("Serial opened");
  digitalWrite(LEDR, LOW);  digitalWrite(LEDG, LOW);  digitalWrite(LEDB, HIGH); // turn the LED yellow
  delay(1000);
  Serial.println("         1         2         3         4         5         6");
  Serial.println("1234567890123456789012345678901234567890123456789012345678901234");
  Serial.println(".........this works fine........................................");
  digitalWrite(LEDR, HIGH);  digitalWrite(LEDG, HIGH);  digitalWrite(LEDB, LOW); // turn the LED blue 
  delay(1000);
  Serial.println(".........this causes a crash.....................................");
  digitalWrite(LEDR, HIGH);  digitalWrite(LEDG, LOW);  digitalWrite(LEDB, HIGH); // turn the LED green 
}
```
Produces this output, and then the crash occurs.
The LED light sequence should be black/red/yellow/blue/green, but what occurs is black/red/yellow/blue and then the crash of 3X long, 3X short blink pattern.

```
10:58:37.047 -> Serial opened
10:58:38.066 ->          1         2         3         4         5         6
10:58:38.066 -> 1234567890123456789012345678901234567890123456789012345678901234
10:58:38.066 -> .........this works fine........................................

```

I tried another Arduino to see if the problem is with the unit, and it happens on the second one as well.  Both of them are Arduino Nano 33 BLE.  My next step will be to break your longer displays into shorter lines, just so I can get the IMU calibrated.

Any insight as to why that crash is occurring would be appreciated.

cdeerinck, 2020-11-15:
I captured this one time (MBed OS crash info).  Not sure exactly what sequence allowed me to capture it:

```
10:11:51.879 -> Serial opened
10:11:51.879 ->          1         2         3         4         5         6
10:11:51.879 -> 1234567890123456789012345678901234567890123456789012345678901234
10:11:51.879 -> .........this works fine........................................
10:13:09.420 -> <.<..=console>tty?nullle@zeroleAklogleBoslogeCoslog_streamDfdoE
10:13:09.420 -> stdinstF
10:13:09.420 -> stdouttG
10:13:09.420 -> stderrtSerial opened
```

I also ran the code on a PC (all previous activity was on a Mac) and got the same results.

cdeerinck, 2020-11-15:
Recognizing that your Github is not the Arduino forum, I opened a post there at: https://forum.arduino.cc/index.php?topic=713904.new and will close this.  But I thought it might be helpful to others that run into the same problem.  FYI, I have fixed and run the Gyro and Acc calibration programs, with lines like this, until I can get to the root of the problem.

`Serial.print(F(" Before calibrating choose the Full Scale (FS) setting and ")); Serial.println(F("Output Data Rate (ODR). The accelerometer and the"));`

Thank you very much for the hard work you put into writings this.  My Gyro is now usable!

closed 2020-11-15 by cdeerinck

FemmeVerbeek, 2020-11-16:
Hi Chuck, 
Its always nice to know that people are using my code. 
Yes, the forum is the best place to get help. I'm not always available reading here. I found a new job in the midst of corona, but it takes an awful lot of time off  :) . 
Good to know that the problem is not in the library and I've never heard of it. I hope you solve it.

Femme.