FemmeVerbeek/Arduino_LSM9DS1 #13: Serial Monitor Freezes during Calibration

rezaabdullah, 2020-12-02:
Serial Monitor freezes during the calibration process.  It seems inside calibration function such as `calibrateAccelMenu()`  the serial monitor freezes.  The output shows `Calibrate Accelerometer Offset and Slope`. It does not run beyond that. The behavior is the same for gyroscope and magnetometer calibration. The `simple` examples work. Only the three calibration code fail. Below is the screenshot
[image]

In the meantime built-in LED blink 3 times with 1 sec delay and 6 times with shorter delay. Not sure if the LED has any significance?

Nejat11m, 2021-01-05:
I have the same issue. Have you find a solution ?

rezaabdullah, 2021-01-06:
I'm using a different library. Since I am using Nano 33 BLE Sense (which has onboard LSM9DS1), I just manually enter all the registers and address.

FemmeVerbeek, 2021-02-01:
I can't reproduce your error. It was running fine using Windows 10 and the build in serial monitor. There were lots of problems in the past with the address of the virtual comport changing after complilation in the IDE. So you had to reconnect every time. But that's a problem inside the IDE.

JorgWieme, 2021-03-15:
Reproducible on Ubuntu 20.04.2 LTS and Windows 10. After looking into it, it's apparently a known issue in [ArduinoCore-Mbed](https://github.com/arduino/ArduinoCore-mbed/issues/85), where writing strings longer than 64 bytes gives this problem. This has been resolved and included in releases 1.3.1 and 1.3.2. When this problem occurred is was running with mbed 1.3.0. I have not been able to change my mbed version due to some other issues but as a temporal solution I simply split the longer strings.