FemmeVerbeek/Arduino_LSM9DS1 #14: Using the library for internal and external LSM9DS1

gswonk, 2020-12-21:
The BLE Sense has an internal IMU. I need an additional one at a different address.
Both SparkFun and Adafruit sell an external version.

Does this library support the alternate address of an additional IMU?

FemmeVerbeek, 2021-02-01:
Somebody used it on an external IMU using a classic Aduino UNO with only 5% of the memory, so yes it can be done. I don't know if he had to modify any internal register addresses.

gswonk, 2021-02-01:
Thank you.

I turns out on the Arduino BLE sense there are two different I2C busses.

Glenn

FemmeVerbeek, 2021-02-01:
I'm not a specialist on the I2C busses. AFAIK that communication is in the wire library.

gswonk, 2021-02-01:
Correct, there are two instances on the BLE sense.

Wire and Wire1.

I do have it partially working with two LSM9DS1.
I can give you the examples code when I finish it.

Glenn

FemmeVerbeek, 2021-02-01:
I'll probably never use it myself, but perhaps for others reading this.

gswonk, 2021-02-06:
Femme,

Thanks for your library. I use it on the BLE Nano sense.

So can you confirm that the accel and gyro always are at the same output
data rate?

I think you say that you can only utilize the accel, which is what I want
to do.

So, I just need to disable or not enable the gyro.

IMU.setGyroODR(0);  // switch off gyro
IMU.setAccelODR(2);  // 50 Hz ODR

I did confirm that there are two I2C interfaces - internal and external.
Use with Wire or Wire1 - or both.

So theoretically, you can use up to three devices (one internal, two
external).

thanks,
glenn

FemmeVerbeek, 2021-02-07:
Hi Glenn,
Correct, and yes that's how to disable the gyro. There are three operational modes 0=off , 1= Accel only , 2= Gyro +Accel, see int getOperationalMode(); 
When both are working they always share the same ODR.

Note that the ODR values in the documentation tables are not acurate, not good enough for navigation purposes. The library measures the actual ODR at every setting change and stores it internally, but this was not designed to serve multiple IMU's. If you need the acurate number per IMU, you must store it yourelf. You can get the values by calling getAccelODR()  getGyroODR() and getMagnetODR()

Femme

gswonk, 2021-02-07:
Femme,

Thank you so much.I owe you a beer.

I am working on a commercial product and am not sure based on high volume
pricing which one is best.
We are working on a proof of concept prototype and the LSM9DS1 was the
easiest to acquire.

Do you have an opinion on which IMU is the best based on accuracy and price.

thanks,

glenn

FemmeVerbeek, 2021-02-07:
No, sorry. The LSM9DS1 is a big seller. If the specs are good enough for you it is probably a good choise. The ODR is a bit disappointing. The makers promised 952Hz, but in reality it is closer to 500Hz. They say you need 2000 for drone flight orientation.

gswonk, 2021-02-07:
Ok, thanks.

I only need 10/second and does not work with the default code.

I know your code works because I blink the LED and it works as stated.

thanks,

glenn

gswonk, 2021-02-07:
Femme,

I was able to get the external LSM9DS1 to work on the NANO33BLE.

Unfortunately, I had to modify the code to force the use of Wire, not Wire1.

I tried declaring IMUexternal separately, but it didn't work.

See the code below.

thanks,

glenn

#ifdef ARDUINO_ARDUINO_NANO33BLE
///LSM9DS1Class IMU(Wire1);
LSM9DS1Class IMU(Wire);

// By default on the NANO33BLE, the code uses the internal I2C bus.
// If needed, also enable using the LSM9DS1 on the external I2C bus.
LSM9DS1Class IMUexternal(Wire);

#else
LSM9DS1Class IMU(Wire);
#endif

FemmeVerbeek, 2021-02-07:
It does not make sense to me. 
The declaration you refer to is in the private section,  so IMUexternal should not be visible outside the library.
Didn't you forget to init the new Object?
The idea of OOP is that you reuse code, not rewrite it. 
The library predeclares and inits one object of the LSM9DS1Class called IMU. 

If you want to use a second chip you should declare it yourself in your sketch.  
e.g.
#ifndef ARDUINO_ARDUINO_NANO33BLE
LSM9DS1Class IMUexternal(Wire);
IMUexternal.begin();  // don't forget to init your object.
...
...

#endif

That should give you two objects 
IMU which was already predeclared connected to Wire1
IMUexternal connected to Wire

gswonk, 2021-02-07:
Femme,

Yes I agree. I just needed a quick fix to keep moving forward on my project.

I will do as suggested and update you when it is working.

I will also provide you with some working code for pitch and roll
processing based on the IMU acceleration.

thanks,

glenn