FemmeVerbeek/Arduino_LSM9DS1 #24: setMagnetODR Error

trexmiller, 2022-05-23:
Upon compilation...

/Users/trexmiller/Documents/Arduino/libraries/Arduino_LSM9DS1/src/LSM9DS1.cpp: In member function 'virtual int LSM9DS1Class::setMagnetODR(uint8_t)':

/Users/trexmiller/Documents/Arduino/libraries/Arduino_LSM9DS1/src/LSM9DS1.cpp:449:12: error: control reaches end of non-void function [-Werror=return-type]
  449 |   magnetODR= measureMagnetODR(duration);    //measure the actual ODR value
      |   ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors

Using earlier version (1f5a19dbfae6aa72236273ad24e0aef7378d1f12) compiles with no errors

FemmeVerbeek, 2022-05-23:
Hi Todd,LSM9DS1 was written as update.You must remove the old version completely from the directory where the libraries are stored.Better ask these questions in the arduino forum. I have little time.FemmeMet vriendelijke groetFemme Verbeek06 2606 0303

Using earlier version (1f5a19d) compiles with no errors

trexmiller, 2022-05-23:
Yeah, I get it. Just that I did remove it completely, but I will check it again.

BTW, we cited your library in our latest paper to come out shortly in ACS, ES&T Water.